JaredC01/Galileo #1: Share Source / Solid Files

vertigo235, 2020-09-16:
Can you please share your solid files / source, the original designer has licensed this as creative commons share alike. 

https://www.thingiverse.com/thing:4223085

JaredC01, 2020-09-16:
> Can you please share your solid files / source, the original designer has licensed this as creative commons share alike.
> 
> https://www.thingiverse.com/thing:4223085

They're already available, if you know where to look.  I will not be posting the source files here until the project is out of beta, as I do not want partially complete / needing revision files to be posted here.  If you need additional info, please send me a DM on Discord.

Thanks,
Jared

vertigo235, 2020-09-16:
Will do, didn't mean to imply anything, was just following breadcrumbs 👍

closed 2020-09-16 by vertigo235