JaredC01/Galileo #2: Missing links to additional models

shift, 2021-01-13:
Would it be possible to get links to the other required STL files required for the extruder, for example the part cooling parts?

Thanks

JaredC01, 2021-01-13:
Hi, all of the additional parts are "stock" to the Voron models, and can be found on the Voron GitHub. Please let me know if you need further assistance.

shift, 2021-01-14:
Which Voron base model are the part cooling parts from, and which variant? I see some similar ones on the 2.1 branch.

JaredC01, 2021-01-14:
Both the 2.1 and 2.2 should have the same parts with the exception of the X carriage. Here's the link for the 2.2 toolhead, including the cooling ducts (use the ducts labeled as V6 / Mosquito).

https://github.com/VoronDesign/Voron-2/tree/Voron2.2/STLs/VORON2.2/Gantry/X_Carriage/Toolhead

For clarity's sake, I may update the readme as you had originally requested, at least with regards to the 2.1/2.2 design.

prdg79, 2021-01-21:
For the Dragon, will the cooling ducts for the Mosquito work?

thiagolocatelli, 2021-04-30:
I have been reading about 2 hours about the Galileo Clockwork and how to assemble and I can't find anything. If you say that all the other parts are "stock", where did Bullrolls take STLs from for the Galileo Clockwork that look very different from what is on the manual? https://www.aliexpress.com/item/1005001947783573.html

closed 2022-07-25 by JaredC01