JaredC01/Galileo #6: Compatibility with Orbiter V1.5 and upcoming V2

buskun, 2021-05-30:
I have read through the README.md and found that there is only mentioned to the original Orbiter not V1.5 or upcoming V2. 
As V1.5 has some of [its internal bearing dimension has been changed](https://www.thingiverse.com/thing:4725897#:~:text=internal%20bearings%20dimension%20has%20been%20changed), I don't want to assume that parts from V1.5 or V2 will compatible with building Galileo.

JaredC01, 2021-05-31:
The changes from the original design to the 1.5 have not changed compatibility with the Galileo Clockwork, as all of the core components are reused (that are designed to work together), however the 2.0 will not be compatible.

buskun, 2021-05-31:
Ok! Thank you for your response. I will close this issue now.

closed 2021-05-31 by buskun