JaredC01/Galileo #13: rear plate for trident mgn12 carriage

RPisarew, 2021-11-28:
Hi,

thanks for the nice mod, but unfortunately the notch of the rear plate does not fit into the trident carriage. It's some millimeters to long.
It would be awesome if you can add a second version of the rear plate for the trident carriage :)

JaredC01, 2021-11-28:
I'm not sure which version you've got downloaded, but Galileo has been adjusted for Trident since before Trident was released. There should be two screw notches on the rear plate for clearance.

RPisarew, 2021-11-28:
Hmm, I have printed the latest version of it.

RPisarew, 2021-11-28:
This section does not fit for me:

[image]

RPisarew, 2021-11-28:
Here is a picture of the printed parts:

[image]

JaredC01, 2021-11-28:
I've just checked the CAD against the Trident carriage, and there's no interference.  Is it possible your M3 screws are too long and aren't seating fully?  The fit is tight, but it does (just) clear.

RPisarew, 2021-11-28:
The fit is perfect, it's just a bit to long. I will remove some material from the end of the notch. But as I said, I have printed all parts yesterday and I downloaded the latest files and the result is in the picture above.

RPisarew, 2021-11-30:
Sorry for the inconvenience, it does fit. I tested it without the screws and expected them to be higher...

closed 2021-11-30 by RPisarew