JaredC01/Galileo #14: Galileo Extruder - Stealthy style?

neophrema, 2021-12-17:
Good evening,
First of all, thank you very much for the current design, I'm building it currently and are quite impressed.
The Voron Team did just release a new toolhead assembly, do you have any plans into customizing the Galileo extruder so both work together? Since I'm only able to - narrowly - use freecad, I think that project would be to big for me.. But maybe I can assist a little.
I think planetary gears are superior to the normal assembly, I would very much enjoy using your extruder with the new fanduct designs.
Thanks!

JaredC01, 2021-12-17:
The Stealthburner cooling system is already fully compatible with the Galileo Clockwork. That said I have some projects in the works as well, so we'll see what happens.

neophrema, 2021-12-17:
Oj, that are great news! I was hoping for compatibility! Also im interested into your work obviously ... you can do a hint whitch quartal/season of 2022 you plan to release? So ill drop by? All the best, have a merry christmas and a good new year!

neophrema, 2021-12-22:
Quick request, sorry that I come back to you.
I had a closer look at the stealthburner:
"Let's hang out and get an early look at Stealthburner!" -- https://yewtu.be/watch?v=Znr-Qy1n5Jg

It appears to me that the bottom Part of the extruder is altered.- you are sure, or do you just estimate that the galileo will be compatible with the new toolhead? Might be that the screws have the same measurements, but the plastic guides/slides seem different. 
Otherwise I might need to adapt it or ... even postpone the planetary gear idea for now..

JaredC01, 2021-12-22:
I have Stealthburner running on my Trident with Galileo as we speak. The Clockwork 2 module requires some changes, but the cooling ducts known as Stealthburner are a drop-in replacement.

pranas, 2021-12-25:
I understand you can only use the new Stealthburner front but you cannot yet use the new X carriage with extruder screws on the front or the new toolhead mounts, correct?

JaredC01, 2021-12-25:
That is correct. The existing Clockwork will not work with the new carriage. The next release of the Galileo Clockwork (GC2) will follow the Afterburner CW2 design and mounting.

TekuSP, 2022-01-16:
I am looking forward to Galileo 2.

kclau, 2022-02-21:
Any news on Galileo Clockwork 2?

phamtuanvt, 2022-02-22:
> That is correct. The existing Clockwork will not work with the new carriage. The next release of the Galileo Clockwork (GC2) will follow the Afterburner CW2 design and mounting.

i'm waiting for Galileo for Stealthburner day by day, will you upgrade it soon?

meanderer-tech, 2022-05-24:
At this time if you want Galileo extruder for the stealthburner you will need to stay with CW1 version of mounts and toolhead because that's what Galileo was designed against.

PhilBaz, 2022-07-25:
Hello, curious to see if you will embark on Galileo Clockwork (GC2) with the RC1 CAD ? Also, I just wanted to campaign to keep the latching mechanism on the left of the extruder similar to CW1/2, rather than hiding it under the new cable shroud.

JaredC01, 2022-07-25:
> At this time if you want Galileo extruder for the stealthburner you will need to stay with CW1 version of mounts and toolhead because that's what Galileo was designed against. for example https://github.com/VoronDesign/Voron-Afterburner/blob/sb-beta/STLs/Stealthburner/Toolheads/revo_six_%26_v6/stealthburner_toolhead_(v6_r6)-rear-cw1-beta6.stl

I do actually have an updated mount for the CW2 carriage, it's just not published as it's a temporary fix.  I will be working on cleaning up and updating my repository here soon and will manage this at that time.

> Hello, curious to see if you will embark on Galileo Clockwork (GC2) with the RC1 CAD ? Also, I just wanted to campaign to keep the latching mechanism on the left of the extruder similar to CW1/2, rather than hiding it under the new cable shroud.

Galileo Clockwork 2 will be a completely different set of parts, but will be "stealthy" to match the CW2 design.  I will also be releasing a CW2 styled Galileo Clockwork 1 for those not wanting to move to Galileo 2.  As far as the base for the design is concerned, I haven't decided yet whether to use RC1 as the base or wait for full release.  Once there's an update, I'll update my repository.

insane78, 2022-08-07:
The stealthburner is finished, voron team has deployed the release. I can't wait for the new Galileo Jared. The mechanics (gears ecc) will be the same as Galileo 1? Thanks for your awesome job mate.

JaredC01, 2022-08-07:
> The stealthburner is finished, voron team has deployed the release. I can't wait for the new Galileo Jared. The mechanics (gears ecc) will be the same as Galileo 1? Thanks for your awesome job mate.

It's at RC1, not full release yet.  That said, I will be using it for the style upgrade for Galileo.  The mechanics will be the same as Galileo 1, but there will be a Galileo 2 version eventually.

PhilBaz, 2022-08-12:
@JaredC01 I understand the new orbiter 2 has different gears and proportions for the shaft ect.... Will the new CW2 Galileo use the original 1&1.5 internals or the new 2.0 internals ???

JaredC01, 2022-08-12:
> @JaredC01 I understand the new orbiter 2.0 has different gears and proportions for the shaft ect.... Will the new CW2 Galileo use the original 1&1.5 internals or the new 2.0 internals ??? 

Galileo 2 is no longer based on the Orbiter internals, it's completely different.

PhilBaz, 2022-08-12:
I guess my question is referring the the statement above.

" CW2 styled Galileo Clockwork 1"

can i assume 1.0/1.5 internals for this?

JaredC01, 2022-08-12:
> I guess my question is referring the the statement above.
> 
> 
> 
> " CW2 styled Galileo Clockwork 1"
> 
> 
> 
> Can i assume 1.0/1.5 internals for this? And that it will work with the new carriage?
> 
> 

Ah, my apologies, yes that will be the case. Just the 1.0/1.5 parts will be compatible.

herrla, 2022-08-18:
@JaredC01 do you have any estimation when the Galileo update will be available? I'm close to finishing the mechanics of my V2.4 build and am currently considering waiting for your update or buying a bmg kit for CW2.

PhilBaz, 2022-08-21:
@JaredC01 

Hi Jared. I'm curious if you felt the Galileo 1 had improved flexible handling over the orbiter 1.5?  In a similar vein, do you think there will be improvements for flexibles handling between Galileo 1-CW2 & 2.0?

JaredC01, 2022-08-21:
> @JaredC01 do you have any estimation when the Galileo update will be available? I'm close to finishing the mechanics of my V2.4 build and am currently considering waiting for your update or buying a bmg kit for CW2.

I'd like to have something working by the end of August, but that's not a firm timeline at this point. I've got a few projects I'm working on currently, so I can't give a firm date.

JaredC01, 2022-08-21:
> @JaredC01 
> 
> 
> 
> Hi Jared. I'm curious if you felt the Galileo 1 had improved flexible handling over the orbiter 1.5?  In a similar vein, do you think there will be improvements for flexibles handling between Galileo 1-CW2 & 2.0?

Galileo / Orbiter used the same core parts, but I'm not sure if the Orbiter 1.5 had as constrained of an output as Galileo does...  that also said, the larger drive gears make flexibles worse in general unless you can constrain the output to as close to the drive gears as possible.

My opinion is that Galileo 1 isn't great for flexibles overall, but the Orbiter 2.0 metal exit DOES make it work quite well with flexibles.  I'm working with LDO on making the metal exit available for the Galileo 1 refresh, as well as Galileo 2.

PhilBaz, 2022-08-21:
@JaredC01 

I have a 1.5 and from a bit of research people say its not well suited for flexibles. 

That is very interesting about getting the metal exit ( I know what you mean) as a part from LDO. Please ask him (if possible) to prevent triangle-labs, Mello, etc putting a shipping price of $8 on it, as is their usual scam.

Also, I saw people were having good results with the 1.0 using PTFE as the exit tube. trimming it to fit as close as possible to the gears, for a constrained exit path. If its not to much trouble perhaps you can include a version that would also allow for the PTFE method. 

Thanks again!

Jomo-desktop, 2022-10-08:
Hi,Thank you for your work. I love the Galileo that you design. I hope to see your Galileo design for cw2 soon. I've been waiting for it for a long time.

biiiink, 2022-10-29:
Just curious if the Galileo 2 design will accommodate PCBs like the hartk stealthburner PCB 
https://github.com/hartk1213/MISC/tree/main/PCBs/Stealthburner_Toolhead_PCB?_ga=2.240057987.1092711658.1666924321-1272581396.1665780136

or this canbus PCB
https://www.aliexpress.com/item/1005004675264551.html?spm=a2g0o.store_pc_home.0.0.36ab224dxw64Ms&pdp_npi=2%40dis%21USD%21US%20%2418.75%21US%20%2415.56%21%21%21%21%21%4021050b1f16620267366495598e9c9d%2112000030063375966%21sh&gatewayAdapt=4itemAdapt

PhilBaz, 2022-11-21:
https://www.youtube.com/watch?v=JLUDLJQXZeU&ab_channel=MisadventuresinMeltingPlastic

23:33

Drop?

JaredC01, 2022-11-21:
> 

Nah, that's still Galileo 1, it's just got a reworked rear plate to fit the CW2 carriage.  Galileo 2 Extruder is getting awfully close to testing though.

tomnz, 2023-02-03:
I'm about to rebuild my Voron 2.4 gantry with MGN12H rail and Stealthburner, and was hoping to throw Galileo in the mix too - a shame to see this project has stalled out? I have all the Galileo hardware ready to go, but was hoping for updates to printed parts. @JaredC01 are you still planning to put something out here?

neophrema, 2023-02-05:
@tomnz hey, there. A few people would be happy to see a release, but please don't be that pushy, shame is the wrong wording here. There is no obligation to release anything. 
give (at least kindness) and you shall receive ;-)
sincerely,

StackingDeezLayers, 2023-03-18:
Hey. Any word on when the Galileo clockwork 2 (GC2) is going to come out? And any chance it will support toolhead PCBs?

JaredC01, 2023-03-18:
Galileo 2 (and the projects associated with it) are just now being tested in a closed group.  The parts, as well as the drive gears are fully custom.

Once I have more info available for release, it will be posted here.  When release comes, the whole GitHub repository will be updated as well.

phamtuanvt, 2023-04-05:
> Galileo 2 (and the projects associated with it) are just now being tested in a closed group. The parts, as well as the drive gears are fully custom.
> 
> Once I have more info available for release, it will be posted here. When release comes, the whole GitHub repository will be updated as well.

does it use the gear set of the V2.0 orbiter?

PhilBaz, 2023-04-05:
> does it use the gear set of the V2.0 orbiter?

He mentions in the quote they are fully custom.

phamtuanvt, 2023-04-06:
> He mentions in the quote they are fully custom.

so my orbiter is useless :(

phamtuanvt, 2023-04-07:
> @phamtuanvt
> 
> https://www.teamfdm.com/files/file/656-super-stealth-orbiter-orbiter-15/
> 
> But we should cut the chit chat here.

what i want is for it to mount SB2209/SB2240

goodwin, 2023-06-08:
no news?

PhilBaz, 2023-06-09:
@goodwin 

Do you really care?! Galileo was just an orbiter mod. Lots of ways to get an orbiter 1 or 2 into a Voron.  Plenty of developments on the extruder scene too. Move on.

JaredC01, 2023-06-09:
> no news?

Galileo 2 is my current focus, and should be released in the next couple of months. As mentioned previously, G2 is a completely new design, not associated with anything Orbiter.

Once G2 is released, I'll shift my focus back to G1 in a CW2 style. There's a few major hurdles on G1 currently, which is why I have paused G1 development.

-Jared

AudiNV, 2023-06-27:
Following :)

rcg94, 2023-08-22:
Peripherally saw this in a builds video, excited to see the bom and test it out myself. Thank you for your work.

MartijnGevaert, 2023-08-22:
> Peripherally saw this in a builds video, excited to see the bom and test it out myself. Thank you for your work.

Link pls

enerbe, 2023-08-26:
hi @JaredC01 
Any recent news? I wanted to upgrade to Stealthburner. Unfortunately, I had already taken apart the extruder before I realized that I could not put it back together because I have Galileo parts. I'm happy to join the beta team, because it's not functional right now.

KungfuPancake, 2025-03-26:
Since this issue is still open: will this be worked on sometime in the future? I have Galileo parts and would like to reuse them, ideally in the form of a CW2 style extruder :)