JaredC01/Galileo #15: Orbiter 2.0 update

pranas, 2021-12-25:
Is there an update coming to take advantage of Orbiter 2.0 internals? The metal exit guide seems like a great improvement for flexibles and the smaller size is always nice.

JaredC01, 2021-12-25:
I will not be using most of the Orbiter 2 parts for the next Galileo design, though the metal exit guide will likely be used, yes.

The main appeal of the Orbiter 2 is a shortened and smaller overall design, which is not needed for the clockwork design.

mmalaiska, 2022-05-13:
> I will not be using most of the Orbiter 2 parts for the next Galileo design, though the metal exit guide will likely be used, yes.
> 
> The main appeal of the Orbiter 2 is a shortened and smaller overall design, which is not needed for the clockwork design.

how is it going with the next Galileo design? really appreciate your work, cheers

closed 2022-07-25 by JaredC01